lf-lang/reactor-uc #35: Avoiding naming collisions

erlingrj, 2024-10-03:
There is a problem that the name of types and structs defined in  'reactor-uc` might collide with user names. I think the only solution to this is to prepend something to our reactor-uc types and basically rename everything...
1. `lf_DelayedConnection`
2. `lfDelayedConnection`
3. `uc_DelayedConection`
4. `ucDelayedConnection`
5. `_DelayedConnection`

What do you think?

edwardalee, 2024-10-04:
I think we want to consistently use CamelCase or snake_case.  snake_case is more common with C code.

erlingrj, 2024-10-04:
Thanks for suggestion. I guess then either:
1. `LfDelayedConnection`
2. `UcDelayedConnection`

Would be right. The naming convention I went for was to Use CamelCase for structs/classes and snake_case for function names and variable names.

tanneberger, 2024-10-04:
yes CamelCase for struct/classes

I prefer the 2.) `UcDelayedConnection`

closed 2024-12-11 by erlingrj