lf-lang/reactor-uc #43: Avoid booleans as struct fields

erlingrj, 2024-10-07:
Good feedback from Edward. A boolean as a struct field wastes 31 bits. I think we can use C's ability to specify bitwidths as follows
```
struct MyStruct {
  bool first: 1;
  bool second: 1;
  int small_int:6
  uint8_t bigger_int:8
  uint16_t biggest_int:16
};
```

This should all fit in one 32bit word

edwardalee, 2024-10-08:
I don't think you can assume the compiler will put these in a 32-bit word.  What I had in mind was more like this:
```
struct my_struct_t {
   int32_t flags;
   ...
}
```
Then have macros like `FIRST(struct)` and `SECOND(struct)` that you use in conditionals.

erlingrj, 2024-10-08:
Then what is the meaning of 'int a:4'? I thought it was exactly for this purpose?

edwardalee, 2024-10-08:
Oh, I don't know.  I'm not familiar with that syntax.