lf-lang/reactor-uc #51: Add logging

erlingrj, 2024-10-10:
It is overdue to implement a logging system. I want it to be simple but flexible. I want
- Specify logging of arbitrary "modules" with e.g. `LOG_REGISTER_MODULE("scheduler")`
- Set verbosity level for each module `LOG_LEVEL("scheduler", LOG_INFO)`
- Set verbosity level for all modules in one go.
- Do logging specific to each module `LOG_ERR("scheduler", "Tried to schedule event into the past %ld", event_time);`

Zephyr has a particularly nice logging system and I want our logging to match nicely to that: https://docs.zephyrproject.org/latest/services/logging/index.html

For other targets we can chose to either:
1. Implement the nice logging system with multiple modules or
2. Just check global verbosity level and dump everything out over printf.

tanneberger, 2024-10-10:
@LasseRosenow is there a nice riot-os logging module the only thing I found is:
https://doc.riot-os.org/core_2lib_2include_2log_8h.html

erlingrj, 2024-10-11:
I am looking at this currently :)

LasseRosenow, 2024-10-11:
> @LasseRosenow is there a nice riot-os logging module the only thing I found is: https://doc.riot-os.org/core_2lib_2include_2log_8h.html

From my knowledge in RIOT it is common to just use printf and then hide Debug prints behind the DEBUG define, which can be overwritten on a per-file basis.

This logging system that you are referencing there is the better alternative to that in my view, but usage of it is somewhat "random".

erlingrj, 2024-10-11:
A first iteration of flexible logging system is now in main

closed 2024-10-11 by erlingrj