lf-lang/reactor-uc #75: Enable selecting which NetworkChannel implementations to add to the build.

erlingrj, 2024-10-18:
We need a way for conveniently selecting which NetworkChannel implementations to add to the build. Currently it is a CMake option which generates a compile def, but this wont work for RIOT since it is Make based. Can we do it purely in C, i.e. in a code-generated file we select this?

erlingrj, 2024-10-20:
Also, setting the size of the event queue and reaction queue is done from CMake, this needs to be done in a build-system agnostic way.

tanneberger, 2024-10-21:
reactor-cpp did this by having a config.hh.in 

https://github.com/lf-lang/reactor-cpp/blob/master/include/reactor-cpp/config.hh.in

This is the [autoconf](https://www.gnu.org/software/autoconf/manual/autoconf-2.68/html_node/C_002b_002b-Compiler.html) way of passing build parameters to your program

tanneberger, 2024-10-21:
I just realized we already have this file.

erlingrj, 2024-10-21:
The config stuff with CMake is nice, but we dont want it tied to CMake because RIOT, and potential other use cases are Make based.

I think `autotools` might be an alternative, or simple script based on a toml file?

erlingrj, 2024-10-24:
- Just rely on compile defs for selecting platform, networkchannels and reaction-queue and event-queue sizes
- Remove config.h.in stuff

closed 2024-10-26 by erlingrj