lf-lang/reactor-uc #84: Implement min_spacing for actions

erlingrj, 2024-10-22:
min_spacing is a optional argument to an action restricting the minimum time delta 
 between two events scheduled on the action.
 
 It was easy to support when we required actions scheduled monotonically. But now that we will support arbitrary schedulings (#71) we need a complete implementation. 
 
 A proposed algorithm looks like this:
 When `lf_schedule` is called with a tag `t`, if there is a min_spacing. We must search through the event queue and guarantee that there are no other events from the same action scheduled in the time-window" [`t-min_spacing`, `t+min_spacing`]. If so, drop the event.
 
 In LF there are actually several policies for handling such an "illegal" schedule:
 - `defer`
 - `drop`
 - `replace`
 
However, is this well-defined if you try to schedule an event "between" two existing events on the event queue? Where do you defer it to? Which of the two events do you replace?

I think for now we will stick with the `drop` policy

edwardalee, 2024-10-22:
If you look at the specification of `min_spacing` here: https://www.lf-lang.org/docs/next/writing-reactors/actions#triggering-time-for-actions, then I think the implementation is very simple.  When you give a `min_spacing`, then monotonic scheduling becomes required.  I think the current implementation in reactor-c takes advantage of this.

erlingrj, 2024-10-22:
OK, I see, in this case it is very simple, I will look at it

closed 2024-12-13 by erlingrj