lf-lang/reactor-uc #126: Fix FlexPRET program size issue

erlingrj, 2024-11-27:
Compiling `examples/flexpret/src/Smoke.lf` results in a binary needing  21KB RAM and 54KB ROM. This is too much, more than twice as much compiling for Zephyr (10KB and 21KB). 

Here is some more info:
```shell
nm build/fp-smoke --size-sort --print-size | tail -30
0000a864 0000016c T _fclose_r
20000018 0000016c D __global_locale
0000a9d0 00000174 T __smakebuf_r
00003730 00000178 T EventQueue_heapify
0000adfc 00000188 T _wcsnrtombs_l
00008118 00000190 t global_stdio_init.part.0
00008a1c 000001a8 T __swsetup_r
00002f24 000001b0 T Scheduler_acquire_and_schedule_start_tag
00001f0c 000001ec T Reactor_validate
20004828 00000200 B contexts
00003ad4 0000020c T fp_exception_handler
00000fdc 00000214 T FederatedConnectionBundle_connect_to_peers
00007d54 00000258 T __sflush_r
00003f90 000002c8 W Reset_Handler
00009650 000002f8 T _free_r
00004988 00000318 t print_integer
200002a8 00000408 D __malloc_av_
20003ea8 000004c0 B __ee_jmp_buf
20004b88 000004c0 b envs
20004368 000004c0 B __ie_jmp_buf
00008330 000004e8 T __sfvwrite_r
20002500 00000540 B main_reactor
0000a1dc 000005bc T _realloc_r
0000276c 000005d8 T Scheduler_run
00005a3c 00000650 T __umoddi3
00004ca4 00000660 t vsnprintf_impl.constprop.0
00005388 000006b4 T __udivdi3
00009948 0000082c T _malloc_r
20002a70 00001438 b scheduler
00006238 000019e4 T _vfiprintf_r
```

erlingrj, 2024-11-27:
Do you have any insight here, @magnmaeh?

magnmaeh, 2024-11-28:
Hmm print and malloc seem to have the most potential for exploring. But the ROM/RAM usage has always been an issue. But here it looks a little like the std lib printf has gotten into the binary. I might be able to look a little into it this evening.

schoeberl, 2024-12-02:
Does reactor-uc even need malloc?

erlingrj, 2024-12-03:
No it doesn't, but I think the FlexPRET SDK uses it for a couple of things