lf-lang/reactor-uc #160: Add support for multiports with widths based on reactor parameters

erlingrj, 2024-12-10:
The following is currently not supported:
```
reactor R(port_width:int=4) {
  output [port_width] out: int
}
```

The reason is that, in the runtime, the port structs are part of the reactor self struct. When we have a multiport we put an array of port structs on the self struct. However, if the width is based on a parameter, then how large should we make the port struct array? It depends where and how it is instantiated.

The solution is likely to put the port structs on the self structs of the *parent* reactors and passed down as arguments.

edwardalee, 2024-12-10:
I guess I don't see why the structs need to be in the parent reactor's self struct. Isn't there a separate instance of the self struct for each reactor instance?  I guess the required memory could be a static array and a static length variable for each reactor instance. The self struct then just gets initialized to point to the array and set the length. But probably you've figured out a better way (or at least one that will work... (I have no idea whether what I'm describing even comes close to solving the problem).

erlingrj, 2024-12-10:
You are right, and I think we are describing the same thing actually!

Currently, the self-struct of a reactor with a multiport looks a little like this:

```c
typedef struct {
  Reactor_R_out out[10];
 } Reactor_R;
```

We would have to change it into

```c
typedef struct {
  Reactor_R_out *out;
  size_t out_width;
} Reactor_R;
```

We would code-generate a constructor for this reactor that looks like this:

```c
void Reactor_R_ctor(Reactor_R *self, Reactor_R_out *out, size_t out_width);
```

My initial proposal was that the array was to be put in the parent of `Reactor_R`, but I see now that this wont work either. Because it could be derived from a parameter arbitrarily high up in the containment hierarchy. 

So where should those arrays live? An option would be to have them as global variables with unique names and then just referenced when we call `Reactor_R_ctor`. A challenge would be that we would need to do an initial pass through the AST and find all multiports and sort out their sizes.

I also see now that this also applies to bank of reactors. As they also appears as arrays on the self structs of the parent reactors.

This actually seems to be a quite hard problem since we decided to not do dynamic memory allocation....