lf-lang/reactor-uc #163: Introduce syntax for specifying max number of pending events on delayed connections and actions

erlingrj, 2024-12-11:
In reactor-uc we statically allocate all events which means that there is a bound on the number of outstanding events that can be buffered for each action and eached delayed/physical connection.

We need a syntax for specifying these bounds.

Proposal 1:
```
reactor R {

```

erlingrj, 2024-12-13:
@tanneberger what do you think?

edwardalee, 2024-12-13:
Without a strong preference, I like:

```
   @max_pending_events=15
   logical action a(0, 10 msec)
```

and

```
@buffer=15
src.out -> sink.in after 100 msec
```

erlingrj, 2024-12-14:
Thanks, I also realize that we need a "policy" for buffer overflows, either we "drop" the event, or we "replace" an existing. We might also need to specify which event to replace, I think either you want to replace to earliest event, the last event, or the event with the closest tag.

closed 2025-01-24 by erlingrj