lf-lang/reactor-uc #176: make test is broken

schoeberl, 2024-12-13:
On my Mac when running `make test` I get:
```
/Users/martin/source/lf-world/reactor-uc/src/platform/posix/tcp_ip_channel.c:15:10: fatal error: 'sys/eventfd.h' file not found
#include <sys/eventfd.h>
```

schoeberl, 2024-12-14:
I found that `eventfd` is a Linux thing, not Posix or general Unix. Therefore, the tests break in MacOS.

erlingrj, 2024-12-14:
That is unfortunate, @LasseRosenow, do you know of any POSIX alternative? I think Zephyr also supports eventfd, but I don't see anything for RIOT? Unfortunately several interested people are using macOS...

LasseRosenow, 2024-12-16:
Ouuuups. I didn't know this is not available in macOS.

For RIOT I personally find it difficult to find a solution that really works everywhere. The POSIX Wrappers in RIOT really are too incomplete.
I think it is worth considering switching to native RIOT sockets if we want to use TCP there. But to be honest the TCP implementation is very bare bones and incomplete.

But I fixed it for macOS switching to socketpairs, which is POSIX compliant.
Should be fixed in #178

PR #178 (2024-12-16): Migrate eventfd to socketpair to support macOS
Fixes #176
closed 2024-12-16 by erlingrj