lf-lang/reactor-uc #205: Add support for specifying the safe-to-assume-absent value for each port and an exception handler for violations

erlingrj, 2025-01-28:
While we are at it, we could try coming up with a more descriptive name than `STP` or `STAA`. Since we do not bother with zero-delay-cycles in reactor-uc the semantics of this is more straight forward.

erlingrj, 2025-01-28:
Since we dont support zero-delay cycles I don't think we need to differentiate between safe-to-process and safe-to-assume-absent, as we do with reactor-c. Currently it looks like this:

```
reaction(in) {=
  // Handle input
=} deadline(100 msec) {=
  // Handle deadline-miss
=} STAA (20 msec) {=
  // Handle the scenario where we assumed that `in` was absent at a tag, but later got a message for that tag
=} 
```

Currently, the keyword used in LF is `STP` (short for safe-to-process) or `STAA` (safe-to-assume-absent). While they are descriptive, it may be a little confusing because it is like an "inversion of control". You specify a STAA value on a reaction in order to let _other_ reactions execute in time. In my mind, it makes more sense to annotate important reactions with a duration which it is willing to wait in order to make sure that earlier events on other ports are seen. We already have an annotation for this, the `deadline`. 

So I propse something like this:

```

reaction(important_input) {=
  // Handle the important input
=} deadline(100 msec) {=
  // Handle deadline violations for the important input
=}

reaction(less_important) {=
  // Handle less important input
=} expired {=
 // Optional handler for the scenario where we receive the less_important input out of order
=}
```

Now I propose that the deadline of 100 msec is what informs the selection of the safe-to-assume-absent parameter for `less_important`. The trick is, of course, what should the value be exactly. Because, if we wait 100 msec and then receive an earlier event, we will violate that deadline. But I still think that using the deadline is

erlingrj, 2025-01-28:
We could also NOT reuse the `deadline` keyword and use another one. E.g. `defer`, `await`, or `grace` (for grace period). It could look like this then:

```
reaction(important_input) {=

=} deadline(100 msec) {=

=} grace(10 msec) 

reaction(less_important) {=

=} expired {=

=}
```

Woule love to hear other peoples thoughts on this: @edwardalee , @tanneberger , @LasseRosenow ?

tanneberger, 2025-01-28:
I would be cautious introducing special syntax for it. 

If we want to realize our goal with annotating coordination protocols on a connection basis, some of these value are potentially unnecessary and confusing.

tanneberger, 2025-01-28:
@erlingrj 

Another point: I don't think it makes much sense to set STAA offsets for reaction bodies.

My suggestion for ptides coordination:

```
reactor Federate1 {
   @port:ptides(STAA=10ms)
   input in: int

   reaction(in) {= 
       printf("Received Value: %i, in->value);
   =} coordination_error {=
       printf("Received a coordination error: %e", err->msg);
   =} deadline(100 msec) {=
       printf("Deadline Violation");
   =}
}

reactor Federate2 {
   output out: int;
   // ...
}

main reactor {
   @federate:ptides(STA=20ms);
   fed1 = new Federate1();
   fed2 = new Federate2();
   
   @connection:ptides
   fed2.out -> fed1.in;
} 
```

I would keep the error handlers generic so that we don't have to introduce new syntax when we implement other coordination protocols. Also, our reaction could be triggered by input ports that use other coordination schemes.

erlingrj, 2025-01-29:
This proposal is about two elements:

1. A way of annotating when we can assume that an input port is absent. (my proposal is using deadline/grace on important reactions)
2. A way of handling out-of-order reception of events (due to 1.) (my proposal is using the `expired` keyword)

I don't agree that this is specific to a particular coordination protocol. Even

edwardalee, 2025-01-29:
I think there might be some confusion here.
STA = X means that at physical time T+X, the federate can assume it knows all inputs with timestamps _less than_ T.

STAA = Y means that at physical time T+X+Y, the federate can assume it knows a particular input _at_ timestamp T.

STAA is not needed if you don't allow zero delay loops, but in that case, you also cannot use STA as defined above.  You need another threshold, let's call it FOO = Z, which means that at physical time T, the federate assumes it knows all inputs with timestamps _less than or equal to_ T+Z.

The exception handler should not be called a coordination error.  There are legitimate uses where it is not an error.  The simulation examples I will describe next week (in the lf-demo repo) are good examples. Any input that requires inputs to be handled in timestamp order relative to other inputs _on the same port_, but doesn't need them to handled in timestamp order relative to other ports, can use the exception handler to handle them normally.

erlingrj, 2025-01-29:
What I have implemented for reactor-uc is identical to STAA, but we don't have any STA. So the semantics is:

STAA = X means that at physical time T + X the federate can assume that it knows a particular input at timestamp T. 

Before advancing to logical time to t, the federate must check if any inputs are unknown at t, and if so, wait for their STAAs, if there are multiple unknown input ports it will wait for the largest STAA value (and of course wake up if any message arrives).

But, I would like to highlight again that I think it makes more sense to annotate high-priority reactions with a duration for which it is willing to wait, and infer STAA values for the other ports. It means that STAA is not a static property of a port, but rather it depends on which events are present at the tag you want to advance to.

erlingrj, 2025-01-30:
After a discussion with Edward yesterday I have two updates:

1. We should not use the term STAA unless it has the exact same semantics as its existing usage. This is not the case because STAA is intertwined with STA, and there is an assumption that when you consider the STAA for port P at t you have already committed to executing tag t and thus you are assuming that port P is known at all tags less than t, STAA is the duration you wait before assuming its absent on tag t.

2. In today's meeting, I will put together some examples to guide the discussion

closed 2025-02-18 by erlingrj