lf-lang/reactor-uc #234: Level Calculation Failes

tanneberger, 2025-03-02:
[image]

[image]

tanneberger, 2025-03-02:
Looks like a problem in the code-generation where for some reason there is one source allocated for this reaction

[image]

tanneberger, 2025-03-02:
https://github.com/tanneberger/fsw25/blob/main/src/SensorTestUnix.lf

erlingrj, 2025-03-02:
Thanks. The problem is that the code-generator says that both instances of GpioPin have 1 source for the input ports, while in reality, only one of them has it. So definitely a bug in the code-generator that I will fix Monday morning :)

PR #235 (2025-03-03): Fix calculation of reaction sources, effects and observers in parent 
Fixes #234 

Also adds another smoke test for contained ports
closed 2025-03-03 by erlingrj