lf-lang/reactor-uc #241: Name collisions

erlingrj, 2025-03-10:
I have observed the following problems:
- Reactor instance named `env` does not work because already field named `env` on the self struct
- State variables and ports cannot have the same name

Both can be solved by prefixing the user-given names with something, e.g.
- state variables: `__lf_state_`
- ports: `__lf_port_`
- child reactors: `__lf_reactor_`
- parameter: `__lf_param_`
- actions: `__lf_action_`

edwardalee, 2025-03-10:
Wouldn't it be better to prefix the internal fields?  The above proposal would require changing all reaction code that references a state variable, for example, using `self->state_name`.

erlingrj, 2025-03-10:
Thanks for taking a look. Yes, changing the state variable name would be very stupid; lets only change the ports, actions, and child reactors. And then bring them into scope with the correct names.